elhep/AFCZ #1: rev 1 errata

gkasprow, 2018-07-21:
- [ ] BSC050N03LSGATMA1CT-ND footprint is wrong - 5x6 instead of 3x3. Mounted CSD17308Q3 instead

gkasprow, 2018-08-21:
- [ ] DNP R626
- [ ] swapped SDA with SCL on Exar 2 programming connector

gkasprow, 2018-10-03:
- [ ] R289 = VCC

gkasprow, 2018-10-31:
- [ ] add caps for 1.2V rail clock inputs.

gkasprow, 2018-12-11:
- [ ] short reset pins 1 and 2 of IC56
[image]

gkasprow, 2019-10-30:
Crystal oscillator frequency, do not use 48MHz, must be 33.33MHz.

gkasprow, 2019-11-08:
- [ ] add pullup on ETH1 RX_CLK (PHY address)
- [ ] short ETH1 CRS  with ETH1 RX_DV

krmac, 2019-11-08:
> 
> 
> Crystal oscillator frequency, do not use 48MHz, must be 33.33MHz.

Correct value is 50 MHz.

filipswit, 2020-06-17:
@gkasprow could you please explain following:
 

-  R289 = VCC

- add caps for 1.2V rail clock inputs.
 

-----------------

> BSC050N03LSGATMA1CT-ND footprint is wrong - 5x6 instead of 3x3. Mounted CSD17308Q3 instead

Transistors (CSD17308Q3 ) are already changed.

>  short reset pins 1 and 2 of IC56

Jumper was added to IC56
[image]

>  add pullup on ETH1 RX_CLK (PHY address)
 short ETH1 CRS with ETH1 RX_DV

Added jumper and pullup:
[image]

filipswit, 2020-06-22:
>  add caps for 1.2V rail clock inputs.

Where should I add caps?

gkasprow, 2020-06-22:
All clock rails in the banks supplied from 1V2

gkasprow, 2020-06-22:
they are clocked from 1V8 and common mode is far above 0.6V which is required by diff SSTL12

filipswit, 2020-06-22:
ok, I haven't seen two clocks in SDRAM banks.

gkasprow, 2020-06-22:
Can you open dedicated issues if you have some doubts? Otherwize we will miss important things.

filipswit, 2020-06-25:
ok

closed 2020-06-25 by filipswit