elhep/AFCZ #12: FMC+ connector

gkasprow, 2019-11-15:
CERN asked:
Concerning the v2 of the board, would it still be possible to makes some small modifications, typically on clocking and up to when ?
 
A question aside, For now we will use the “FMC” versions of the AFCZ with 6.4 Gsps ADC FMCs but we were initially aiming at 10 Gsps which would require a FMC+ connector with minimum 10x12.5 Gbps lanes. Would it be of interest for you to have such a version of the AFCZ? In other words an AFCZ with one FMC+ connector with minimum 10 serial links routed and a standard one.

filipswit, 2020-06-16:
Do we make any modifications with FMC+?

gkasprow, 2020-06-16:
It's a lot of work.
We cannot increase the number of layers.
The suitable connectors are already in the CERN library. You can check if it is feasible without a serious redesign of the board.

filipswit, 2020-06-17:
Mechanically it's possible;) But routing additional differential links doesn't look pleasantly.

gkasprow, 2020-06-17:
LEt's ask CERN about the importance of this modification.

kaolpr, 2020-06-22:
CERN:
> t is not a priority for this release. We prefer limiting the changes too as for now we will use the links as they are.

I think the issue can be closed for now.

closed 2020-06-22 by kaolpr