elhep/AFCZ #15: Power budget

kaolpr, 2020-06-04:
Power budget must be created taking into account the power requirements of FMC and RTM standards. 

The technique of creating is not important, however, please attach all files to the project repository. 

Please include values from standards and power supply margin on every rail in percent of value required by the standard.

gkasprow, 2020-06-04:
Use Analog devices Power CAD tool and incude all LDOs, regulators, etc https://www.analog.com/en/design-center/ltpowercad.html

filipswit, 2020-06-05:
Do we have any approximations for ZU7 power consumption per rail? And maybe logs from XRs? 
That would be useful in case of research for new power supply solution - #18 .

gkasprow, 2020-06-05:
What is wrong with existing power solution?
Just repace the Exar chips with older 7725 which work just fine.

kaolpr, 2020-06-08:
#18 is closed. We're staying with XR-s. 

@gkasprow I'm bumping a topic of power usage estimation for ZU7. If none exists we'll try to use Xilinx PE. Either way, if #31 is to be implemented we'll have to do it for ZU11.

marmeladapk, 2020-07-16:
Bumping again.