elhep/AFCZ #16: FMC VADJ

kaolpr, 2020-06-04:
If placement constraints allow, for greater interoperability `VADJ` should be adjustable according to the FMC standard, if not possible consider at least isolation of `VADJ` from `P1V8` and add MMC-controlled switch to allow switching `VADJ` off if level requested by FMC is different than 1.8V.

filipswit, 2020-06-05:
Adjustable VADJ would cause adjusting ZU7 supply. I think for now only switch is achievable.

gkasprow, 2020-06-05:
With recent FPGA families, the only possible VADJ is 1.2V , 1.5V and 1.8V.
1.2V and 1.5V signaling is used only in DDR memories so far. Some open hardware FMC modules are recently adopted to support 1.8V LVCMOS signaling. 1.8V does not break interoperability with LVDS supplied from 2.8 and 3.3V. The modules I developed during last two years already support 1.8V signaling. IMHO since we cannot support 2.5V, supporting 1.2 and 1.5V is not worth the effort.

kaolpr, 2020-06-08:
Shall we support disabling VADJ if incompatible FMC is detected by MMC? I'm thinking of the situation when an IC is to be powered by 3.3 and 2.5, but 2.5 turns out to be 1.8. It seems a little bit contrived example though.

gkasprow, 2020-06-08:
1.8V won't break 1.5V FMC board in most cases. It can simpy not enable VADJ when incompatible card is detected. One must make sure that missing VADJ won't harm the SoC  - the same rail may be used to supply other things.

filipswit, 2020-06-08:
I may try to add switch for both VADJ rails.

gkasprow, 2020-06-08:
That's not necessary. You don't want the FMC to steer disabled VADJ bank. Some FMCs are supplied from 3V3 and output voltage to the LA pins. In such a way they may break the FPGA.
It's better if MMC reads the FMC EEPROM and if it finds that voltage range is wrong, it won't enable SoC and FMC supplies at all. It will then blink red led indicating fault.

gkasprow, 2020-06-08:
That's why it is essential to supply the FMC EEPROM from management power.

filipswit, 2020-06-16:
So, finally we leave it as is.

gkasprow, 2020-06-16:
Yes, but we should add an issue to the MMC

kaolpr, 2020-06-26:
It turns out that it was already mentioned on openMMC repo: https://github.com/lnls-dig/openMMC/issues/85.
We'll need to make sure it is implemented in MMC port for AFCZ.

closed 2020-06-26 by kaolpr