elhep/AFCZ #17: FMC inrush current

kaolpr, 2020-06-04:
Some FMCs can draw a significant current when FMC P12V is turned on. Gerd Kotzian at CERN made a measurement with FMC card of one of the leading manufacturers and obtained the following results:
[image]
[image]
These were obtained when powering board with power connector. When board is placed in mTCA crate overcurrent limits of the power supply prevents AFCZ from powering up correctly.

One of tested solutions was adding 100 nF capacitor between gate and source of the MOSFET switch for FMC power supply line to limit inrush current. The following was observer by Gerd after application of the fix:
[image]
Inrush current is significantly lower in this case, so probably this can be applied as a solution.

filipswit, 2020-06-05:
Loks like already fixed. 
[image]

kaolpr, 2020-06-05:
@gkasprow Did you added this capacitor or CERN added another in parallel?

gkasprow, 2020-06-05:
this value needs to be bigger than originally used. It's better to ask CERN about it. Maybe they added another 100nF and 200nF did the job.

gkotzian, 2020-06-09:
Please note that a capacitor was added between gate and _DRAIN_, leaving the 100nF between gate and source unchanged. A value of 100nF did the job. Further details on this subject can be found [here](https://www.onsemi.com/pub/Collateral/AND9093-D.PDF).

LTSPICE circuit simulation:
[image]
[image]

filipswit, 2020-06-16:
@gkotzian thanks for correcting.

closed 2020-06-16 by filipswit