elhep/AFCZ #18: Misbehaving XR77129

kaolpr, 2020-06-04:
XR77129 happened to misbehave on prototype units - it drove some lines with invalid voltages, though settings seemed fine. Replacing with pin-compatible XRP7724 solved the problem. Using XRP7724 instead of XR77129 can be considered as a solution. 

However XRs caused quite a lot of problems in several projects. Thus if #15 (power budget) allows a transition to [*Sayma* style power supply](https://github.com/sinara-hw/Sayma_AMC/releases/download/v2.0.1/Sayma_AMC.PDF) may be a better option. It proved to be out-of-the-box working solution.

gkasprow, 2020-06-05:
XR7724 work reliably. 77129 caused a lot of issues in two other projects. Replacement of the supply section is major rework and adds a risk

kaolpr, 2020-06-08:
Some time ago we've discussed that with you at CTI and then these two ways were considered. However, I'm totally fine with XR7724.

closed 2020-06-08 by kaolpr