elhep/AFCZ #19: FMC Address Pins

kaolpr, 2020-06-04:
Both FMC slots have the same GA address. Connect `GA1` of FMC2 to `3V3PAUX`.

gkasprow, 2020-06-05:
why? We have I2C switch so why not to keep the same addressing?

kaolpr, 2020-06-08:
Indeed ANSI/VITA 57.1 in rule 5.82 states that GA codes should be unique along with IPMI I2C bus. However, I see no reason not to differentiate the slots. It feels like more natural to me. After all, these GA pins are there for a reason and one can expect GA codes to be different in two slots, though specification does not impose that.

Is there any particular reason to do it in the way it is currently implemented?

gkasprow, 2020-06-08:
So far the only reason was that some ICs used on FMCs do not support 4 unique addresses that can be implemented in an easy way. Some ICs, especially from Texas have address pins that support multi-level logic. When building a multi-level I2C mux tree it's nice to have the same chips under the same addresses, but I'm not very fixed on that idea.

filipswit, 2020-06-09:
How about adding resistors to GA1 of FMC2?

gkasprow, 2020-06-09:
that's OK. But as I said, I personally don't really care if we do it one or another way.

gkasprow, 2020-06-09:
I used geographical addressing in 16-FMC carrier box for JET experiment. It was somehow helpful to identify which FMC was which :)

kaolpr, 2020-06-10:
@filipswit GA1 of FMC2 should be connected to `3P3VAUX` according to Rule 5.83.

closed 2020-06-16 by filipswit