elhep/AFCZ #21: Problematic Ethernet Switch

kaolpr, 2020-06-04:
AFCZ is equipped with a bit problematic Ethernet switch - Marvell 88E6321-A0-NAZ2I000. It is a very versatile IC what comes with the price of difficult configuration.

Some problems with integration with Linux on Zynq were reported (@wzab?). However, it seems that they can be mitigated by writing configuration to EEPROM and not supporting switch reconfiguration from Linux.

Using some simpler switch could be considered provided a chip with required set of ports can be found:
* 100Mbit MII interface for LPC1776
* 2x SGMII for AMC P0/P1
* 1x / 2x RGMII for Zynq (**do we need 2x RGMII for Zynq PS?**)

gkasprow, 2020-06-05:
I spent a lot of time looking for a similar switch. This was the only one. So far the main issue is wrong MDIO configuration where ZYnQ fights with MMC for access. I tried it with Devkit. The basic configuration works just fine. Linux needs to be convinced not to touch it and treat it as a dummy PHY chip.
Anyway, we are using it with Linux and it works.
If we skip the MMC, it can be replaced by ordinary PHY like on the Marble design.

wzab, 2020-06-06:
I have got it working with Linux. The problem however is lack of support in U-Boot. That impairs development, as it is not possioble to use the most efficient scenario with kernel+initramfs downloaded via TFTP.

kaolpr, 2020-06-08:
@wzab If we could configure switch from EEPROM would it be ok for U-Boot? Will it solve the problem you report?

gkasprow, 2020-06-08:
We can also configure it from MMC.

kaolpr, 2020-06-29:
We'll stick to the current solution.

closed 2020-06-29 by kaolpr