elhep/AFCZ #29: DESY recommendation class D1.4 for RTM connector

kaolpr, 2020-06-05:
- [ ] Rename at least nets to be compatible with D1.4 recommendation (preferably also ports of the RTM_Connector sheet)
- [ ] Change *NOT USED* to *ASSEMBLY VARIANT*, precisely highlight signals that are available in assembly variant
- [ ] `J31 / 8-10 / ab` do not follow the recommendation, evaluate the possibility of keeping current `RTM_FPGA_CFG` as an assembly option while routing proper signals to these ports

filipswit, 2020-06-09:
I can rout differentially these signals J31 / 8-10 / ab, but from recomendation they should be GTP or CC.
[image]

The RTM_CLK (which from recomendation is AMC_CLK ) actually goes to cross-bar
[image]

Another one is is "RTM_ADC_DAC_SYNC.SYNCOUT11" and it is possible to route it to cross-bar, but please pont which input I can use to make it useful

gkasprow, 2020-06-09:
The plan for RTM in AFCZ was to eventually replace the Sayma AMC. But for the moment it's not important

kaolpr, 2020-06-17:
Up to my understanding, GPTs are just transceiver clocks. In Ultrascale architecture, it is possible to also output signal from the REFCLK pin pair. See #37.

Regarding compatibility - I think generally it would be better to stick to the standard. Even if it's not that widespread. Especially if it does not cost us that much (like f.e. in case of AFCv4).