elhep/AFCZ #32: Add EEPROM for assembly configuration

kaolpr, 2020-06-05:
Additional I2C EEPROM for storing assembly configuration would be nice. Let's assume that position identifier is 15b + 1b for presence, so every position occupies 2B. It's hard for me to say now how many elements are affected with variants, but I guess that 8kB EEPROM will be more than enough.

If the power budget on P3V3_MP allows it would be the best supply. Otherwise up to designer convenience.

filipswit, 2020-06-05:
Does it look fine for you?
24AA64T-I/MS

gkasprow, 2020-06-05:
Use something that is already used on the board. Make sure there is no address conflict. Add its address to the address table.

gkasprow, 2020-06-05:
I would connect WP pin to MMC or FPGA with pullup/pulldown that makes write protection active by default. This is to make sure no one writes to it by chance.

kaolpr, 2020-06-08:
There is an ongoing upgrade of AFC with [the similar issue](https://ohwr.org/project/afc/issues/85). It should be resolved by tomorrow. Please use the same solution to increase unification between projects.

gkasprow, 2020-06-08:
Let's add 24C256, 24C512 or similar chip that can hold the entire configuration xml file.
Connect Write Enable to MMC and pull it up to make writing blocked by default.