elhep/AFCZ #37: Clocking tree

kaolpr, 2020-06-17:
There are several issues regarding clocking tree for AFCZ:

* it is quite complex what makes it not really straightforward to use,
* CERN has some requirements, that seem not to be met at the moment (will be discussed in the following comments to this issue)
* some unification with [AFC4 clocking tree](https://ohwr.org/project/afc/issues/43#note_116367) may be considered (taking into account presence of SoC)

As a starting point for the discussion let's take a diagram (hopefully accurate) of the current AFCZ clocking tree:

[image]

gkasprow, 2020-06-17:
I would get rid of:
Si5324 - it will be replaced by DDMTD anyway
SY89540 and replace it by assembly option as in AFCv4
TCLKC and TCLKD optionally as an assembly option or not use at all.

kaolpr, 2020-06-17:
Rev 1
[image]
If one wants to modify this diagram [yEd source is here](https://github.com/elhep/AFCZ/files/4792167/AFCZ1v1_Clocking_rev1.zip).
.

filipswit, 2020-06-22:
May I start implementing Rev.1 to schematics?

kaolpr, 2020-06-22:
It would be great if @gkasprow would approve it, but I think you can start implementing it.

You can swap clock matrix ports according to the layout constraints.

gkasprow, 2020-06-22:
Looks good to me. Go ahead.

kaolpr, 2020-06-26:
CERN reports that for their application a simultaneous use of:

* GTP8-11_CLK connected to clock matrix and
* MGT226 fed with WR clock.

Suggested resolution: connect CLK7 to MGT227 REFCLK0 with an assembly option to GTP4-7_CLK_IN/OUT.

filipswit, 2020-06-26:
@kaolpr Could you please edit the diagram?

kaolpr, 2020-06-26:
Yes, but after 18. I've uploaded the wrong file in the zip (PNG instead of diagram source) and I don't have access to that file right now.

kaolpr, 2020-06-29:
[image]
Diagram source: [AFCZ1v1_Clocking_rev2.zip](https://github.com/elhep/AFCZ/files/4844874/AFCZ1v1_Clocking_rev2.zip)

kaolpr, 2020-07-03:
- [ ] Invalid CLK11 connection
[image]
- [ ] Swapped FMC2_GBTCLK0_M2C with FMC2_GBTCLK1_M2C (connected to wrong MGT clocks)
- [ ] Don't think we really need control buffer to AMC_TCLK and uFL connector. They can be always-on.

There are also couple of missing assembly options, but these are still to be defined.

Why do you use `Undefined` values of resistors in some places?

filipswit, 2020-07-03:
> Swapped FMC2_GBTCLK0_M2C with FMC2_GBTCLK1_M2C (connected to wrong MGT clocks)

Can we leave it as is? It's already routed on pcb? If not I'll swap it.

kaolpr, 2020-07-06:
Yeah, we can keep it swapped.

kaolpr, 2020-07-06:
One remark for routing - clock matrix ports can be assigned in a way that would be easier to route.

commit 6835d5c2 (2020-07-02): Fixed: #37, #40
commit 1f490d1d (2020-07-07): Fixed: #37, #41, #42, #44, #45, #47
closed 2020-07-07 by filipswit