elhep/AFCZ #41: Prevent MMC resets when opening terminal for controlling serial ports on Linux

kaolpr, 2020-06-29:
See https://ohwr.org/project/afc/issues/134. Suggest implement it the same way.

kaolpr, 2020-07-08:
This circuitry was decided to be greatly simplified in AFC4:
[image]
Also having this approach implemented SW2 does not seem to be needed anymore.

filipswit, 2020-07-08:
Did you have any issues with TXB level translator? Can I leave them or should swap to SN74AVC2T245?

filipswit, 2020-07-08:
Actually we can skip SW2. Is this "bypass" needed? It goes to SW2.
[image]

gkasprow, 2020-07-08:
this switch was used to enable operation outside of MTCA crate

commit 6b7877c0 (2020-07-09): Fixed: #41, #50, #51, #52
closed 2020-07-09 by filipswit