elhep/AFCZ #45: Unify M-LVDS with AFCv4

kaolpr, 2020-06-29:
It's still being clarified but should be finished very soon. Issue added not to forget about that.

- https://ohwr.org/project/afc/issues/87
- https://ohwr.org/project/afc/issues/131

kaolpr, 2020-07-03:
M-LVDS implementation for AFC4 is ready. We can unify. @gkasprow do you know any arguments against?

filipswit, 2020-07-03:
Is pulldown for DriverEnable enough or i should add the translator with FPGADONE as enable?

filipswit, 2020-07-06:
Is this configuration fine for you?
[image]

kaolpr, 2020-07-06:
Regarding pull-down - it may work if PL is configured to be floating during configuration. On AFCZ default PUDC_B config is high, what makes pull-ups disabled during configuration. Therefore it seems that putting pull-downs on driver enable lines should be enough.

filipswit, 2020-07-06:
I haven't find it in spec. MLVDS lines are used unidirectional, 4RX lines and 4 TX, right? 
Because in AFC4.0 all drivers and receivers are connected.

kaolpr, 2020-07-06:
In mTCA every port is defined as a TX/RX pair. According to mTCA 4.1 REQ 6-4 AMC may implement receiver, transmitter or both on every Rx and Tx pair of the ports 17 to 20. 

There are 3 options:
* If LINK.IO8 and PRI_UART were moved from bank 88 we could implement it in AFC4 fashion. This is the preferred solution. 
PRI UART can be moved to Bank 87, and RTM IO is LVDS, so it can be moved to Bank 87, 66 or 68.
* We can connect 1D-1R, 2D-2R ... as it was done in the v1 but there is only one receiver enable signal what effectively would force us to have single direction on all RX lines.
* We can leave it as it is having unidirectional TX/RX lines.

filipswit, 2020-07-06:
So will try to do AFC4 solution. In case of routing problems will leave as is.

closed 2020-07-07 by filipswit