elhep/AFCZ #50: USB JTAG

filipswit, 2020-07-08:
In AFC4 this circuit is greatly simplified
[image]

Simplify it or leave it as is?

gkasprow, 2020-07-08:
remove DNP resistors. IC16 may not be needed.

filipswit, 2020-07-09:
Redesigned to that:
[image]

kaolpr, 2020-07-10:
If SN74AVC2T245 were already introduced, maybe we could consider replacing TXB0304 with those? We don't need a bidirectional level translator here and TXB0304 seems to be a bit vulnerable to layout (glitches from reflections), what we've painfully experienced with FMC TDC card.

gkasprow, 2020-07-10:
Yes, do it. The simpler, the better.

filipswit, 2020-07-10:
Previously I haven't checked space (there will be 4 ICs instead of 2), but i looks fine.

gkasprow, 2020-07-10:
Just copy it from Sayma/Kasli

closed 2020-09-14 by filipswit