elhep/AFCZ #54: Could we change J11 to USB C type connector?

marmeladapk, 2020-07-09:
It's more popular than micro-b 3.0.

kaolpr, 2020-07-10:
Definitely!

gkasprow, 2020-07-10:
are you sure it fits below the panel?

kaolpr, 2020-07-12:
Indeed it may be a bit problematic. USB-C connectors are 3.15 mm high. Looking at 2-15 figure of AMC Spec. maximum high from the center plane (that is in the middle of PCB) should not exceed 3.6 mm. It leads to the conclusion that for AFCZ upper limit of component hegiht is 3.6-0.8 = 2.8 mm.

kaolpr, 2020-07-12:
But maybe we could use a mid-mount USB-C connector, f.e. [this one](https://pl.mouser.com/ProductDetail/JAE-Electronics/DX07B024JJ3R1600?qs=vFWH4%252BYVDKuy2UDF2xBbVQ%3D%3D)? It should fit then.

filipswit, 2020-07-13:
Yes, mid-mount USB-C will fit.

marmeladapk, 2020-07-14:
What do you think about changing J7 also to type-C for consistency?

gkasprow, 2020-07-14:
Good idea, but it would require lane switch.

filipswit, 2020-10-30:
Changed both to USB-C

gkasprow, 2020-10-30:
beware with USBc - the signals are swapped on the other side of the connectors :) It also needs resistors for proper detection.

filipswit, 2020-10-30:
You mean: SSCTX and SSCRX, right? ;)
I've placed 56k PU for CC1 and PU placeholder for CC2, because we don't use RX2/TX2

filipswit, 2020-12-02:
@gkasprow could you please verify if everything is fine with USB-C connectivity on J7 and J11 connectors?

gkasprow, 2020-12-02:
It is not fully USB-C because the HS lines are not swapped after rotation of the connector. If we accept it than fine. We should not call it USB-C because it is still USB3.0. To be fully compilant, we should use [HD3SS3220](https://www.ti.com/product/HD3SS3220?utm_source=google&utm_medium=cpc&utm_campaign=asc-null-null-GPN_EN-cpc-pf-google-wwe&utm_content=HD3SS3220&ds_k=HD3SS3220&DCM=yes&gclid=Cj0KCQiAk53-BRD0ARIsAJuNhpsznlZ5ZS0cbJrFtFN5-SJ7eQCsFMPgauqUWeZQ7rHKEvqVhLPMzR0aAsrGEALw_wcB&gclsrc=aw.ds) or similar

filipswit, 2020-12-11:
Unfortunately due to mechanical issues, we need to revert both connectors to starting variant. FMC panel overlap on mid mount USB-C connector.

closed 2020-12-22 by filipswit