elhep/AFCZ #59: Unify SN74AVC2T245RSWR (only 2 pcs on AFCZ) and TXB0304RUTR

marmeladapk, 2020-07-14:
Could we unify SN74AVC2T245RSWR (only 2 pcs on AFCZ) and TXB0304RUTR? We could save a BoM line.

Price per port is comparable, however I don't think that having part with direction auto sensing is safe on JTAG lines so I would lean towards using SN74AVC2T245RSWR.

AFAICT no signal on TXB0304RUTR used in AFCZ is bidirectional so we don't need direction sensing.

gkasprow, 2020-07-14:
definitely, we should avoid auto-direction sensing transceivers when possible.

marmeladapk, 2020-07-16:
Besides, TXB0304RUTR datasheet states that:

> If pull-up or pull-down resistors are connected externally to the data I/Os,their values must be kept higher than 20 kΩ to ensure that they do not contend with the output drivers of the TXB0304.

On JTAG lines we use 10k pullups.

gkasprow, 2020-07-16:
be careful with auto-sensing level translators and pullups/puuldowns. They can bring to unexpected issues. Just avoid such translators when possible

kaolpr, 2020-07-20:
When replacing this:
[image]
please keep MMC master to maintain compatibility with AFCv4.

closed 2020-09-14 by filipswit