elhep/AFCZ #62: DTR and RTS lines on MMC console

marmeladapk, 2020-07-14:
- AFAICT when FTDI is powered it actively pulls up DTR and RTS lines - this means that IC65 will pull down reset line to MMC unless one opens console and switches DTR and RTS lines
- btw DTR and RTS have pullups on FTDI side and pulldown on MMC side

Could we either implement circuit from Sayma or at least add switches to disconnect DTR and RTS lines? Or use XNOR instead or XOR gate (there still may be a problem when non-uniform pullup will cause short spike on reset line, so I'd be for an option of disconnecting those signals without soldering).

filipswit, 2020-07-15:
It's a bug, there shouldn't be XOR.

filipswit, 2020-09-11:
Placed AND gate - #63

closed 2020-09-11 by filipswit

tprzywoz, 2020-10-30:
Why do we use AND gate here instead of XNOR like on AFC? There shouldn't be a problem with spikes because we use RC filter.