elhep/AFCZ #78: PS has disabled access to flash when P3V3MP is low

marmeladapk, 2020-07-16:
[image]

Is that necessary? I would assume that when P3V3MP is down then Exars should be powered down by lack of enable from MMC.

gkasprow, 2020-07-16:
there is no scenario of using the board without P3V3MP

gkasprow, 2020-07-16:
the disable circuit makes sure that there is no latchup when P3V3MP is on and P3V3 is still off.

marmeladapk, 2020-07-17:
How does it protect from latchup? If P3V3 is off then OE is low regardless of P3V3MP.

gkasprow, 2020-07-17:
The role of T22 was to disable both switches when the MMC was disabled. The original switch configuration routed not used FLASH to MMC, while active FLASH was connected to the SoC.
The 3V3 can theoretically be enabled without P3V3_MP and thus cause the parasitic supply of MMC MCU.

marmeladapk, 2020-07-17:
So the other way around - protects from latchup if P3V3 is enabled and MP is disabled. 

> The 3V3 can theoretically be enabled without P3V3_MP and thus cause the parasitic supply of MMC MCU.

Do we need to worry about this scenario? You said earlier:

> there is no scenario of using the board without P3V3MP

so could we get rid of this circuit or would you rather keep it?

gkasprow, 2020-07-17:
Get rid of it

Pozdrawiam,
G.Kasprowicz

closed 2020-09-11 by filipswit