elhep/AFCZ #103: TCK is pulled up and down

marmeladapk, 2020-07-28:
TCK is pulled up on FPGA_JTAG and down on AMC_JTAG.

AMC spec:

> REQ 6.42 JTAG supporting AMC Modules shall provide a 10 KΩ pulldown for TCK to Logic Ground.

JTAG spec:

> Stored-state devices contained in the test logic shall retain their state indefinitely when the signal applied to TCK is stopped at 0. 
> Stored-state      devices      contained in the test logic may retain their state indefinitely when the signal applied to TCK is stopped at 1. 

Ultrascale UG:

> TMS and TDI have default weak internal pull-up resistors, as specified by the IEEE Std 1149.1, as do TDO and TCK.

It seems that pulldown on TCK should be the default state as it is safer state in JTAG spec. Anyway, pulldown on AMC_JTAG.TCK drives input of IC1 which drives FPGA_JTAG.TCK, so pullup here accomplishes nothing. Anyway TCK has weak pullup inside the FPGA.

On the other hand, when programmer is connected, both platform cable and JTAG-HS3 have pullups on TCK.  [Additionally](https://www.xilinx.com/support/documentation/data_sheets/ds593.pdf#G1307701):

> The output drivers are enabled only during cable operations; otherwise, they are set to high-Z between operations. Xilinx design tools actively drive the outputs to logic 1 before setting the respective buffer to high-Z, avoiding the possibility of a slow rise-time transition caused by a charge path through the pull-up resistor into parasitic capacitance on the target system.

So when either programmer is connected it ensures that TCK is high. When disconnected on powered board it will cause TCK to go low. Not the end of the world, but R76 doesn't change anything here. I'd remove it.

marmeladapk, 2020-08-06:
Additional info from UG583:

> Place 4.7 kΩ pull-up resistors on the TMS, TCK, and TDI lines.

This seems conflicting with other specs.

filipswit, 2020-09-14:
R76 removed.

closed 2020-09-18 by filipswit