elhep/AFCZ #147: Reducing BoM count

marmeladapk, 2020-08-06:
There are multiple types of components which share the same value:

- 100 nF 0402 capacitors: 16V X5R and 50V X7R
- 10 nF 0402 capacitors: 16V X7R and 25V X7R
- 22 uF 0805 capacitors: 16V X5R and 10V Y5V
- 100 Ohm 0402 resistors: 0,0625W and 0,1W

marmeladapk, 2020-08-06:
Can C235 and C236 be changed to a single 2.2nF capacitor? These are the only capacitors of this size an 2.2nF is found elsewhere in the project.

marmeladapk, 2020-08-06:
R497 is used only once - perhaps voltage divider could be changed to use values already in project?

marmeladapk, 2020-08-06:
R542 should is used only once but it will probably dissapear after GPIO expander unification.

marmeladapk, 2020-08-06:
R593 is used only once but is a part of voltage divider which value is important.

filipswit, 2020-09-18:
> Can C235 and C236 be changed to a single 2.2nF capacitor? These are the only capacitors of this size an 2.2nF is found elsewhere in the project.

In DS is 2.2nF, so I chaned to it.

> Can C235 and C236 be changed to a single 2.2nF capacitor? These are the only capacitors of this size an 2.2nF is found elsewhere in the project.

Changed to 1k+2k2

- [ ] 100 nF 0402 capacitors: 16V X5R and 50V X7R

- [ ] 10 nF 0402 capacitors: 16V X7R and 25V X7R

- [ ] 22 uF 0805 capacitors: 16V X5R and 10V Y5V

- [ ] 100 Ohm 0402 resistors: 0,0625W and 0,1W

- [ ] 10 uF: 0603 X5R and 0805 X7R

- [ ] C852, 47uF is single, could be unified with C899 (different size and 10% vs 20%)

- [ ] Change R108 (SOC_SUP) to match R518, R527, R650 (R2512_0R_JUMPER)

closed 2020-10-29 by filipswit