elhep/AFCZ #155: changelog

gkasprow, 2021-04-15:
@filipswit please provide a changelog. Only critical updates.

filipswit, 2021-04-20:
Major changes:
- Added Zynq PL I2C access #35

- Clocking tree changes, removed crosspoint switch - SY89540UMY  #37

- Changed MMC connector to [ftsh-110-01-l-dv-007-k](https://www.samtec.com/products/ftsh-110-01-l-dv-007-k), #46

- Direct connection of Si57x to PL #149

- Access to Flash memory from PL #150

- Sysmon pins connected to I2C bus # 138

- Power monitors changed to INA3221 #94

- Changed M-LVDS driver #45

- Changed SD card translator to TXS02612RTWR

- JTAG chain redesigned #25

- RTM connector signals changed to be compatible with D1.4 recommendation #29

- Changed Ethernet connector to distinguish from JTAG #26

filipswit, 2021-04-21:
spreadsheet with pinout differences between v1 and v2 
[AFCZv1v2diff.xls](https://github.com/elhep/AFCZ/files/6349398/AFCZv1v2diff.xls)